tasmota/mgos32-to-tasmota32 #97: Shelly Mini1G3 Safeboot after migration

m-f-g, 2025-01-20:
Hello,

i have flashed 5 Shelly Mini1G3 without issues.
Number-6 startet after "start migration" in safeboot-Tasmota-14.4.1 

OTA upgrade failed.

What can i do now ?

thanks
bg
m-f-g

Jason2866, 2025-01-20:
When OTA upgrade fails the only way is to flash via serial the tasmota32-c3 factory image

Jason2866, 2025-01-20:
As I wrote in the repo readme, idk why sometimes the OTA convert fails. Never managed to reproduce the fail. I can't fix something I can't reproduce.

m-f-g, 2025-01-21:
Thanks for the info. 

Yes I have read the readme, I know the risk ;-)
so I have to look how to serial flash the mini ...

i flashed all minis one by one in a row. 
the last one maybe 20 hrs later ... 

thanks.
bg
m-f-g

nholloh, 2025-01-26:
@Jason2866 I am able to reproduce it with one of my devices. According to the log, there seems to be a write error due to a non-empty bit.

```
Bits are not empty. Write operation is forbidden.shelly1minig3-e4b32328d88c 1423 400.734 1 0|shelly_update.cpp:448   EFUSE write error bit[2]: 5636shelly1minig3-e4b32328d88c 1424 400.735 1 1|shelly_ota_source_rp:25 leftover RPC source 0x3fcc7ff4shelly1minig3-e4b32328d88c 1425 400.750 1 2|shelly_notification:208 Event from sys: {"component":"sys", "event":"ota_success", "msg":"success", "ts":1737899914.79}
```

  I am attaching the full log as well in case I missed something.

[shelly1minig3_updatelog.txt](https://github.com/user-attachments/files/18550317/shelly1minig3_updatelog.txt)

Jason2866, 2025-01-26:
@nholloh Well that is new and i have not seen before. Is it the version with Matter support from Shelly? If yes, it is not possible to OTA convert the device. Even more there is a high chance that none of the Matter enabled can be flashed via serial with an other firmware than Shelly.

**Never** buy a device which is supporting Matter from manufacturer when a 3rd party firmware should replace the original firmware. The chance this is impossible is close to 100%

nholloh, 2025-01-26:
Yeah I also just figured I should have read the entire post before. So just to clarify, my shelly is not converting at all, so not going into tasmota safeboot mode. Since this may be part of the "some shelly devices are not converting properly" issue, I am replying here. If you would like me to create a separate issue, let me know.

Regarding Matter Support: Its a standard Shelly 1 Mini Gen3. According to Shelly's website, all Gen3 devices can be updated to support Matter, however the 1 Mini Gen3 does not list Matter support on the website and I do not see any matter related config in the web UI.

Here's some device info, if that helps.

```
{
  "web_id": "241010173203-679fcca9-HEAD",
  "deviceInfo": {
    "name": null,
    "id": "shelly1minig3-e4b3231a6a88",
    "mac": "E4B3231A6A88",
    "slot": 0,
    "key": "xxxx",
    "batch": "2430-Broadwell",
    "fw_sbits": "01",
    "model": "S3SW-001X8EU",
    "gen": 3,
    "fw_id": "20241011-114456/1.4.4-g6d2a586",
    "ver": "1.4.4",
    "app": "Mini1G3",
    "auth_en": false,
    "auth_domain": null
  }
```

I have some older Gen3 which I purchased in early 2024. I was able to migrate them to tasmota without issues. The newer model reads v0.1.0 B2445 on the PCB (OTA fails, error see above); the older model is v0.1.0 B4223 (OTA successful).

Jason2866, 2025-01-26:
@nholloh Thx for providing the infos. But without having such a "problematic" device in hands for investigations it is highly unlikely to come to conclusions what needs to be changed if possible.

nholloh, 2025-01-26:
@Jason2866 I see two options to move forward: You could point me towards some guide or info which could help me find the issue myself _or_ I could ship one of my shellys to you, whichever you prefer.

Jason2866, 2025-01-26:
@nholloh Both options are possible.
1) Just look and investigate, makes assumptions, do changes with your assumptions, there is no guide for.
2) You can send me a device. But there is no guarantee i find the reason for and even when there is maybe no solution possible (whatever the reason is). **I don't send devices back.**
Contact me at [Tasmota Discord](https://discord.gg/Ks2Kzd4) https://discord.gg/Ks2Kzd4 when you want to send me the device. My nick is the same

rossi75, 2025-04-30:
> [@nholloh](https://github.com/nholloh) Both options are possible.
>
> 1. Just look and investigate, makes assumptions, do changes with your assumptions, there is no guide for.
> 2. You can send me a [...]

Hi Jason,
could you find out something? I am facing exact the same issue...

Jason2866, 2025-04-30:
@rossi75 did not have time for investigations