tasmota/mgos32-to-tasmota32 #98: Shelly Dimmer 0/1-10V PM Gen3

benni1390, 2025-02-12:
Hello,

I just tried to flash the Shelly Dimmer 0/1-10V PM Gen3 and used the firmware from Plus10V as a test. However, only a JavaScript error appears on the frontend. I have attached the device log as JSON: [diagnostics-shelly-dimmer 0_1-10v pm gen3-debug-log.json](https://github.com/user-attachments/files/18768890/diagnostics-shelly-dimmer.0_1-10v.pm.gen3-debug-log.json)

[image]

[image]

Shelly Knowledge Base: https://kb.shelly.cloud/knowledge-base/shelly-dimmer-0-1-10v-pm-gen3

The CPU seems to already be supported by Tasmota: https://tasmota.github.io/docs/ESP32/#esp32-c3

Best regards

benni1390, 2025-02-12:
A little further on, I checked out the project and built a test firmware for the new Gen3 dimmer.

```sh 
ShellyDimmer0110VPMG3=( Dimmer0110VPMG3 )
for i in "${ShellyDimmer0110VPMG3[@]}"
do
    generatezip $i "tasmota32c3.bin" "fs-512.img" 2
done
```
I've attached the new device log. 
Error could be: 
`shelly_ota.cpp:261 Aborting update: backend error while processing data: -1 (-10: Cannot relocate nvs)`

Jason2866, 2025-02-12:
Does the device has already got updates? If yes please provide the Shelly link to the update package for this device. Without at least this package there is no chance to build a valid convert zip.

benni1390, 2025-02-12:
Yes, there are updates. The device should have been delivered with 1.4.3, and 1.4.4 is currently installed. The url for the updates is: https://updates.shelly.cloud/update/Dimmer0110VPMG3

```json
{
  "stable": {
    "version": "1.4.4",
    "build_id": "20241011-114458/1.4.4-g6d2a586",
    "url": "https://fwcdn.shelly.cloud/gen2/Dimmer0110VPMG3/5d7903f428c7d3fdd469af844276feafea265e706863eda4e47b6887718f2047"
  },
  "beta": {
    "version": "1.5.0-beta2",
    "build_id": "20250203-144331/1.5.0-beta2-gbf89ed5",
    "url": "https://fwcdn.shelly.cloud/gen2/Dimmer0110VPMG3/4cc01f37efbf57467c8d331a5be44266d80b072f7efd09b2d98025534daa3873"
  },
  "time": 1739379021
}
```

Jason2866, 2025-02-12:
Thx, at first sight. Your setup is wrong. The device has a 1M fs partition.

Jason2866, 2025-02-12:
Shoot in the dark. Try this

```bash
ShellyDimmer0110VPMG3=( Dimmer0110VPMG3 )
for i in "${ShellyDimmer0110VPMG3[@]}"
do
    generatezip $i "tasmota32c3.bin" "fs-1024.img" 3
done
```

benni1390, 2025-02-12:
Thank you, the topic with Tasmota and custom firmware is completely new to me, there is still a lot to learn.

ok, there is some progress. 
Built the new firmware with your config, works. Flashing the Tasmota firmware also worked. I am currently at step 8.
Should I just select any device to replace the locked bootloader step?

[image]

Jason2866, 2025-02-12:
Use Shelly S1G3. Will do the job. Settings are wrong but the bootloader will be replaced.
After the safeboot partition convert process it is safe to play around.

benni1390, 2025-02-12:
Unfortunately, it didn't work. The following happened:
I took the config for the Shelly S1G3, Shelly restarts, but does not come back online in the same wifi. 
Instead it provides a wifi, when connecting I get the ip: 192.168.33.9 with the gateway 192.168.33.1, but no frontend.

[image]

```sh
> curl -i http://192.168.33.1/
HTTP/1.1 404 Not Found
Content-Length: 9
Server: ShellyHTTP/1.0.0
Connection: close

Not Found%
```

Jason2866, 2025-02-12:
you can try:

- Download the update package for your device

- Connect to the Shelly AP, let's assume you get the IP 192.168.33.42

- In a second console, listen with netcat to an arbitrary UDP port, e.g.  nc -u -l 56789

- Run curl -X POST -d '{"id":1,"method":"Sys.SetConfig","params":{"config":{"debug":{"udp":{"addr":"192.168.33.42:56789"}}}}}' http://192.168.33.1/rpc to let the Shelly firmware send debug messages via UDP

- In the folder, where you downloaded the ZIP, run python -m http.server to serve this folder via HTTP

- Run curl "http://192.168.33.1/ota?url=http://192.168.33.42:8000/<name_of_the_package>.zip" to trigger the firmware installation

Wait a bit, there should be progress messages coming in via UDP, check the nc console

Afterwards, you can access the Shelly Web UI again.

benni1390, 2025-02-13:
Thanks, I successfully managed to flash the official Shelly firmware 1.4.4 again with your instructions. The Shelly Web UI is accessible again.

benni1390, 2025-02-15:
Unfortunately, I also don't know why the convert process failed.
There were no manual reboots. As far as I remember, there should have been an automatic reboot after setting the Wi-Fi credentials.

I wanted to test it again:
**Starting point:** Original Shelly firmware:

- Upload and flash: mgos32-to-tasmota32-Dimmer0110VPMG3.zip
- Device reboots
- The Tasmota UI is available, and the device is already connected to Wi-Fi without requiring reconfiguration – ?
- Continuing with the guide, Step 8: I used the Shelly S1G3 for the configuration again.
- Step 9: Successful
- Step 10: Successful
- Step 11: Successful
- Test reboot via UI: Successful
- Full reboot without power: Successful

Tasmota seems to be running now, but as expected, with the wrong configuration. As a test, I used the JSON of the Shelly Plus 0-10V Dimmer, but without success.

```json
{"NAME":"Shelly Plus 0-10V Dimmer","GPIO":[288,0,0,0,192,0,0,0,0,0,0,0,0,0,193,1,0,0,0,0,0,32,416,0,0,0,0,0,4736,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
```

- The device was reset again through the UI and Tasmota is now available again.

Qowy, 2025-02-22:
I am guessing the gen3 DALI dimmer will be the same as this right? Tasmota has support, so only the flashing is the question.

neel-m, 2025-04-03:
It failed for me because I must have given it the wrong password for my Wi-Fi network.  It thought it was fine, but ended up with no IP address.  After the reboot I got the "Not Found" on the default web page.  It took some work, but I was able to recover back to the Shelly 1.4.4 FW.  It was not easy.  This step leaves something to the imagination
`Download the update package for your device`
This is one way to find the right package.
1. go here: https://[updates.shelly.cloud/update/MiniPMG3](https://updates.shelly.cloud/update/MiniPMG3)
2. Figuring out the right string to use for the device was a challenge, not sure where the list of those names is kept
3. https://fwcdn.shelly.cloud/gen2/MiniPMG3/3e8a12d902ff7635412b872d574573437490369d69dbc43553ccf20bd1f51a48 is the 1.4.4-g6d2a586 version of the firmware

I was unable to get the curl POST to get logging to work.  It said Bad Request.  Since this was just for logging, I decided to move on.

I first misunderstood which package I was supposed to use and tried using the tasmota package, which didn't work.  Once I found the magic to get the Shelly FW I needed (they do not make it easy) the process worked fine.

Now on to converting to Tasmota.