tasmota/mgos32-to-tasmota32 #99: Connection timeout to AP after flashing

jakeycrx, 2025-02-21:
Hi,

I've flashed a few Shely devices without any issues, but I have a 1PM mini Gen 3 that I cannot not connect to the AP after flashing. The AP comes up on Wi-FI, I can connect to it and ping it, but it just times out when I try and connect to 192.168.4.1 in my browser.

Any ideas?

Jason2866, 2025-02-21:
AP name is "tasmota-xxx"?

jakeycrx, 2025-02-21:
Yes, it was showing as Tasmota. I managed to connect just after I posted. I entered my Wi-Fi details, but on reboot it was running as a Shelly again. I can't add it via the Shelly app again now though. I'll keep trying.

jakeycrx, 2025-02-22:
I gave up, I think it's bricked. I've flashed another 1PM Mini Gen 3 and also a PM Mini Gen 3 with no issues. 

I guess you can't power cycle if you're struggling to connect to the AP?

Jason2866, 2025-02-23:
> I guess you can't power cycle if you're struggling to connect to the AP?

Yes, if the replacement of the bootloader fails, (idk why this seems to happen sometimes), power cycle is the way to a soft brick. The device can be recovered by serial flashing Tasmota factory image

Jason2866, 2025-02-23:
When Shelly AP comes up you can try this https://github.com/tasmota/mgos32-to-tasmota32/issues/98#issuecomment-2654908281

JFHCFC, 2025-02-24:
5 devices Mini 3 Gen 3 (S3SW-001X8EU), 

1 OK
4 Fails ...

how i did it:
1. Devices has 1.1.99-minig3prod1
2. Update to 1.3.3
3. Update to 1.4.4
4. Upload file mgos32-to-tasmota32-Mini1G3
5. After update AP tasmota-xxx presented
6. Logged into tasmota, in web chose WiFi, enter password , save and ... that's it ...

In all devices Tasmota webgui shows correct ip (dhcp from wifi), but in 4 devices webgui is from new ip is urreachable .... i try connect many times from diffrent pc (and android phone) and webbrowsers

so i turn off devices ...
after boot (i read is soft brick...), again shows AP tasmota-xxx, but can't connect
after another boot shows AP Shelly-xxx, after connect (192.168.33.x), ping to 192.168.33.1 is ok but webgui show "not found"

i saw pinout to uart flash buy my soldering skills are very poor, most likely I damaged the pads

jakeycrx, 2025-02-24:
> When Shelly AP comes up you can try this [#98 (comment)](https://github.com/tasmota/mgos32-to-tasmota32/issues/98#issuecomment-2654908281)

Could you go into a little bit more detail for this method please? Unfortunately, I'm running Windows. I have downloaded Zenmap which is supposed to have netcat.

jakeycrx, 2025-02-24:
I tried the update without running netcet. I could only find a copy of 1.3.3. It seems to attempt the update:

[image]

But I get an error from the python http server:

[image]

This is the contents of the zip file I found:

[image]

jakeycrx, 2025-02-24:
> Sorry, can't help with Windows. Don't use.

Do you have any ideas about the python error?

JFHCFC, 2025-02-27:
I saved a few switches

I soldered 4 pads (3.3V, TX, RX, GND)
[image]
[image]

https://tasmota.github.io/install/
to put the device into programming mode, you need to short-circuit GPIO9 and GND (before inserting the converter into USB)

with ota uploading mgos32-to-tasmota32-Mini1G3 always crashes at this point, then restarts and you can enter the shelly webgui again (I also tried with fm beta 1.5.1b1 uploaded) 

[image]