tasmota/mgos32-to-tasmota32 #100: Shelly 1PM Gen3

michitimmi, 2025-02-27:
Hello,
I want to flash my Shelly 1PM Gen3 with Tasmota. But there is no firmware for that.

[image]

Shelly Knowledge Base:   https://kb.shelly.cloud/knowledge-base/shelly-1pm-gen3

Best regards

JFHCFC, 2025-02-28:
You need to download coresponding zip
https://github.com/tasmota/mgos32-to-tasmota32/releases

then (in shelly) settings ->  firmware -> upload zip and flash

read
https://github.com/tasmota/mgos32-to-tasmota32

"⚠️ Currently some devices fail sometimes to convert. Reason unknown. ⚠️"

michitimmi, 2025-02-28:
Which zip file should I use? Is the S1G3 the right file for my S1PMG3?

JFHCFC, 2025-02-28:
Shelly PM Mini Gen3 - Power monitoring and internal temperature monitoring from HAA Manager
MiniPMG3.zip

Shelly 1PM Gen3 - Switch, power monitoring and internal temperature monitoring from HAA Manager with 75°C overheat protection
S1PMG3.zip

Jason2866, 2025-02-28:
No, there is no matching convert zip for Shelly 1PM Gen3 
The internal device name needs to match. `S1PMG3` The zip package from the Shelly PM Mini Gen3 will 100% not work.

michitimmi, 2025-03-05:
Will there be a Tasmota version for the Shelly 1PM Gen3?

Jason2866, 2025-03-05:
@michitimmi  You can serial flash Tasmota via the Tasmota WebInstaller. Tasmota will work.
There is just no Shelly OTA convert zip

michitimmi, 2025-03-05:
Which version should I use?
Can I use the version for 4MB flash?

Jason2866, 2025-03-05:
yes, if you use the Tasmota WebInstaller just select Tasmota. That's all. No need to think about chip or variant. The installer selects the correct one.

Jason2866, 2025-03-13:
Found the update link https://updates.shelly.cloud/update/S1PMG3

michitimmi, 2025-03-14:
I flashed the serial port to Tasmota, and it works. I'm just having trouble with the template for the 1PMG3. There isn't one available yet.

Jason2866, 2025-06-26:
Don't can't the spend time for this at the moment. Don't hold your breath. Buy a serial adapter.