tasmota/mgos32-to-tasmota32 #102: Add Shelly 2 PM Gen 3

Mirodin, 2025-03-13:
Please add support for Shelly 2 PM Gen 3.
They appear to be using the same chipset as Shelly 1 Gen 3.

https://kb.shelly.cloud/knowledge-base/shelly-2pm-gen3

[image]

Jason2866, 2025-03-13:
Sorry, there is no update package available yet. Without the needed setup for an OTA zip can not be done. If you have an link to an original complete firmware zip or to an update zip please post.

Mirodin, 2025-03-13:
Is this the link you were asking about (though it states gen2)?
https://fwcdn.shelly.cloud/gen2/S2PMG3/504311728dd8c380df171bd737105e2ab83878bde783331de0fd9ad23665f0c3

Found it via Shelly update endpoint: https://updates.shelly.cloud/update/S2PMG3

Jason2866, 2025-03-13:
Do not update to firmware v1.5.x. Idk if this firmware adds matter support. If this is the case the device will be more secured and flashing non Shelly firmware will be impossible. Even with serial flashing.

Jason2866, 2025-03-13:
I will have a look the next days to add a convert OTA zip package

Mirodin, 2025-03-13:
> Do not update to firmware v1.5.x. Idk if this firmware adds matter support. If this is the case the device will be more secured and flashing non Shelly firmware will be impossible. Even with serial flashing.

Yeah, I read the warning in the readme but thought the archive might help you in creating the conversion zip.

Jason2866, 2025-03-13:
Thx for the archive! It makes it possible to build a convert package

RavenSystem, 2025-04-11:
I think 1.5.x makes changes over partitions, again.

1.6.x adds Matter support for Gen3, currently in a beta stage. Probably, it uses eFuse to block even wired-flash permanently.

Jason2866, 2025-04-11:
Thx, for the info @RavenSystem  again a new setup needed for v1.5 until v1.6 is on any device, which will finally stop  the show. Thought v1.5 already brings Matter support