tasmota/mgos32-to-tasmota32 #105: Shelly Plus 1PM Mini Serial Flash Problem After "MgOS32 to Tasmota32" Failed

jvincze84, 2025-04-25:
Hi All,

I have a Shelly Plus 1PM mini module.  (SNSW-001P8EU)

I have read this [https://templates.blakadder.com/shelly_plus_1pm_mini.html](https://templates.blakadder.com/shelly_plus_1pm_mini.html)

First I have tried the OTA flash method, as the link above suggested.  

I could access the web interface on http://192.168.4.1, but after I tried to set up wifi connection, I got Successfully Connected message, but the IP address was 0.0.0.0. 

I checked my router DHCP leases, but I faild to find the device. I restarted the device. 

After this point I cannot access my device.

I decided to flash with serial connection. 

Details of the device:

```
sudo python3 esptool.py --port /dev/ttyUSB0 flash_id
esptool.py v4.7-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 54:32:04:ab:9d:b4
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

I tried the following binaries:

* tasmota32.bin
* tasmota32c3.bin
* tasmota32.factory.bin
* tasmota32c3.factory.bin

None of worked. 

I tried with the webinstaller (https://tasmota.github.io/install/) and with epstool.py.
Both of them give me success message, but after the flashing I can't find the Wifi AP. 
(With the web installer I tried not to specify custom image, as well).

Every help would be appreciated.

Jason2866, 2025-05-20:
Tasmota WebInstaller will choose the correct firmware for flashing. After flash disconnect from PC. Reassemble the device and connect to AC mains. It will work!

jvincze84, 2025-05-20:
> Tasmota WebInstaller will choose the correct firmware for flashing. After flash disconnect from PC. Reassemble the device and connect to AC mains. It will work!

Hi,
Thank you very much for the reply.
Finally I figured out that was soldering problem. :(
https://github.com/arendst/Tasmota/discussions/23343#discussioncomment-12971037

Thanks,
JVincze84