tasmota/mgos32-to-tasmota32 #108: Update step gets write error for shelly 1 mini gen 3

XobrekX, 2025-06-13:
I just tried multiple times to update my shelly and the update always states that it is successful, but after the reboot of the shelly the normal UI just comes up and no tasmota AP. 

Soldering is not my strong suit, so I'd love to do this OTA Update. 

In the logs at the end, there is something about a missmatch in a signature, followed by some write errors.
My shelly is running on 1.4.0, i've received them yesterday. 

I appended the whole log file, but this is the part I am curious about. 
`
{
    "ts": 1721994423.318,
    "level": 2,
    "data": "shelly_ota.cpp:771      Update succeeded, will reboot, uba=3, cd=0/ct=0 s",
    "fd": 1
  },
  {
    "ts": 1721994423.32,
    "level": 2,
    "data": "shos_system.c:60        Rebooting in 3000 ms",
    "fd": 1
  },
  {
    "ts": 1721994423.366,
    "level": 2,
    "data": "shelly_notification:208 Event from sys: {\"component\":\"sys\", \"event\":\"scheduled_restart\", \"time_ms\": 2982, \"ts\":1721994423.32}",
    "fd": 1
  },
  {
    "ts": 1721994423.57,
    "level": 2,
    "data": "shelly_ota.cpp:182      OTA status success, 100 %",
    "fd": 1
  },
  {
    "ts": 1721994423.572,
    "level": 1,
    "data": "shelly_update.cpp:416   FW signatures: want 07 got 00",
    "fd": 1
  },
  {
    "ts": 1721994423.573,
    "level": 0,
    "data": "efuse                   Bits are not empty. Write operation is forbidden.",
    "fd": 1
  },
  {
    "ts": 1721994423.575,
    "level": 0,
    "data": "shelly_update.cpp:443   EFUSE write error bit[1]: 5636",
    "fd": 1
  },
  {
    "ts": 1721994423.576,
    "level": 0,
    "data": "efuse                   Bits are not empty. Write operation is forbidden.",
    "fd": 1
  },
  {
    "ts": 1721994423.578,
    "level": 0,
    "data": "shelly_update.cpp:443   EFUSE write error bit[2]: 5636",
    "fd": 1
  },
  {
    "ts": 1721994423.579,
    "level": 1,
    "data": "shelly_ota_source_rp:25 leftover RPC source 0x3fcc92f8",
    "fd": 1
  },
  {
    "ts": 1721994423.6,
    "level": 2,
    "data": "shelly_notification:208 Event from sys: {\"component\":\"sys\", \"event\":\"ota_success\", \"msg\":\"success\", \"ts\":1721994423.56}",
    "fd": 1
  }
`

[diagnostics-shelly-mini1g3-debug-log.txt](https://github.com/user-attachments/files/20722670/diagnostics-shelly-mini1g3-debug-log.txt)

Jason2866, 2025-06-13:
Which convert zip is used (release version number and date)?

XobrekX, 2025-06-13:
I am using this one: https://github.com/tasmota/mgos32-to-tasmota32/releases/download/v14.1.1/mgos32-to-tasmota32-Mini1G3.zip

Jason2866, 2025-06-13:
For Shelly fw v1.4.x you have to use https://github.com/tasmota/mgos32-to-tasmota32/releases/tag/v13.4.2

latest is for Shelly fw 1.5.x and will not work with older Shelly fw versions

XobrekX, 2025-06-13:
And I was not fast enough to download the logs.

Jason2866, 2025-06-14:
If you want to flash esphome you have to solder. There is no OTA way from Tasmota to esphome. Tasmota is using safeboot partition scheme. esphome is not compatible too.

XobrekX, 2025-06-19:
Hey there,
I used this 3d printed thingy to flash my shellys and it works like a charm. Thank you for helping out and answering my questions.

By the way, i was able to flash a shelly mit fw 1.6 (the matter one) via this method without any problem. Maybe this info helps with your future development.

agross, 2025-07-21:
> For Shelly fw v1.4.x you have to use [v13.4.2 (release)](https://github.com/tasmota/mgos32-to-tasmota32/releases/tag/v13.4.2)

That bricked my Mini1G3 with Shelly 1.4.0 firmware as well.

> I thought for everything lower than 1.4.x, the 13.x version should be installed.

Seems like the readme was right after all.